RobotWebTools/webrtc_ros #27: Does not stream video in Firefox

neuromorph, 2016-02-11:
I am capturing images with gscam and publishing to /v4l/camera/image_raw.
The viewer page in webrtc_ros streams the video successfully in Chrome but fails to do so in Firefox. 
Setup is Firefox 43 on Ubuntu 14.04 with ROS Indigo.

**Below is console log from webrtc_ros_server_node:**

[ INFO] [1455192094.350708497]: Waiting For connections
[ INFO] [1455192107.572766937]: Handling Request: /
[ INFO] [1455192107.655739952]: Handling Request: /index.js
[ INFO] [1455192107.758730185]: Handling Request: /list_streams.json
[ INFO] [1455192113.334219726]: Handling Request: /viewer?subscribe_video=ros_image:/v4l/camera/image_raw
[ INFO] [1455192113.426479677]: Handling Request: /adapter.js
[ INFO] [1455192113.536064545]: Handling Request: /viewer.js
[ INFO] [1455192113.536542770]: Handling Request: /webrtc_ros.js
[ INFO] [1455192113.756397701]: Handling Request: /webrtc
[ INFO] [1455192113.756828746]: Handling new WebRTC websocket
[ INFO] [1455192113.857939429]: Creating WebrtcClient
[ WARN] [1455192113.860424613]: Unexpected codec: ISAC/48000/1 (105)
[ WARN] [1455192113.860583408]: Unexpected codec: PCMU/8000/2 (110)
[ WARN] [1455192113.860657394]: Unexpected codec: PCMA/8000/2 (118)
[ WARN] [1455192113.860774019]: Unexpected codec: G722/8000/2 (119)
[ERROR] [1455192113.876667095]: webrtc: unable to set thread priority
[ERROR] [1455192113.876861994]: webrtc: unable to set thread priority
[ERROR] [1455192113.881432877]: webrtc: Built-in AEC not supported on this platform
[ERROR] [1455192113.883172457]: webrtc: Built-in AEC not supported on this platform
[ INFO] [1455192114.406173914]: Starting ROS subscriber
**[ WARN] [1455192114.569979025]: Socket::OPT_DSCP not supported.
[ WARN] [1455192114.570082556]: SetOption(5, 0) failed: 0
[ WARN] [1455192114.570481765]: Socket::OPT_DSCP not supported.
[ WARN] [1455192114.570559140]: SetOption(5, 0) failed: 0
[ WARN] [1455192114.571796324]: No transport found for content "audio".
[ WARN] [1455192114.571929992]: Failed to set up BUNDLE
[ WARN] [1455192114.572584114]: RTX SSRCs configured but there's no configured RTX payload type. Ignoring.
[ WARN] [1455192114.578913526]: webrtc: (rtp_packet_history.cc:43): Purging packet history in order to re-set status.**
[ERROR] [1455192114.579159039]: webrtc: unable to set thread priority
[ WARN] [1455192114.586278397]: webrtc: (rtp_packet_history.cc:43): Purging packet history in order to re-set status.
[ WARN] [1455192114.586493940]: webrtc: SetRenderDelay(0): Invalid argument.
[ERROR] [1455192114.586685976]: webrtc: unable to set thread priority
[ERROR] [1455192114.586810586]: webrtc: unable to set thread priority
[ERROR] [1455192114.586919066]: webrtc: unable to set thread priority
**[ WARN] [1455192114.590257575]: Unexpected stream: default
[ WARN] [1455192114.619640227]: SSLStreamAdapterHelper::Error(ContinueSSL, -1,1)
[ WARN] [1455192114.619882927]: SSLStreamAdapterHelper::Error(ContinueSSL, -1,1)
[ERROR] [1455192115.584626665]: webrtc: GetDelayEstimate() failed to locate channel (error=8002)
[ERROR] [1455192116.585064738]: webrtc: GetDelayEstimate() failed to locate channel (error=8002)
[ERROR] [1455192117.585974481]: webrtc: GetDelayEstimate() failed to locate channel (error=8002)**
...
Any inputs to get it to work in firefox will be appreciated

neuromorph, 2016-02-12:
Hi @mitchellwills , could you pls provide some pointers to debug this issue? In firefox, the viewer page loads but does not display video/image , instead the video part is blank. I also tried using the latest adapter.js from webrtc.
Thanks

Skylion007, 2016-04-04:
This might be the issue? https://hacks.mozilla.org/2015/02/webrtc-requires-perfect-forward-secrecy-pfs-starting-in-firefox-38/

PR #30 (2016-04-11): Fixes issues with Firefox.
Attempts to fix #27 and also fixes audio issues on Chrome.
commit 259f2e79 (2016-04-11): Fixes issues with Firefox. (#30)

* Fixes issues with Firefox.

Attempts to fix #27 and also fixes audio issues on Chrome.

* Updated viewer.html to rely on local copy of adapter.js again
closed 2016-04-11 by rctoris