RobotWebTools/webrtc_ros #31: Latest Chrome Update Breaks WebRTC support

Skylion007, 2016-05-24:
The Latest Version of Chrome seems to have caused this library to stop working. i have been unable to see video through any other browser as well. Firefox does not work, neither does Edge or Safari. As of the latest Chrome update, Chrome no longer seems to work either. As such, this library is no longer functional especially considering how Chrome is forced to autoupdate. The audio function does appear to still work but remote streams fail to function.

Skylion007, 2016-05-25:
Easiest fix would be to upgrade to WebRTC50, but the backend changes are breaking it. I suggest we pull the changes from [this fork](https://github.com/mitchellwills/webrtc_ros/commit/299f08d04c8ca921f12b594a0687f999df8b542b) to at least get WebRTC version 45 working and upgrade it from there.

Skylion007, 2016-06-05:
Good news! The latest version of Chrome seems to fix whatever the previous version broke! We should probably update the WebRTC library to ensure this doesn't happen again though.

roehling, 2019-01-16:
Fixed for now with #34 
The newer WebRTC APIs would require a serious overhaul of the existing source code, and I don't have the time right now.

closed 2019-01-16 by roehling