RobotWebTools/webrtc_ros #36: Catkin Build Error (WebRTC junit-jar)

clungzta, 2017-08-14:
Hello,

I am attempting to install this package on Ubuntu 16.04 ROS Kinetic.

I cloned your repo into my catkin_ws/src directory and proceeded to run `catkin build`

The build process seemed to be succeeding for some time, then this error occured:

```
Errors     << webrtc:cmake /home/alex/catkin_ws/logs/webrtc/build.cmake.000.log                                                                                                                            

src/third_party/junit-jar (ERROR)
----------------------------------------
[0:03:56] Started.
[0:03:56]

________ running 'git init --bare /home/alex/catkin_ws/src/webrtc_ros/webrtc/src/third_party/_gclient_gittmp_junit-jar_NoZJm' in '/home/alex/catkin_ws/src/webrtc_ros/webrtc'
[0:03:56] Initialized empty Git repository in /home/alex/catkin_ws/src/webrtc_ros/webrtc/src/third_party/_gclient_gittmp_junit-jar_NoZJm/

________ running 'git -c core.deltaBaseCacheLimit=2g clone --no-checkout --progress https://chromium.googlesource.com/external/webrtc/deps/third_party/junit --template=/home/alex/catkin_ws/src/webrtc_ros/webrtc/src/third_party/_gclient_gittmp_junit-jar_NoZJm /home/alex/catkin_ws/src/webrtc_ros/webrtc/src/third_party/_gclient_junit-jar_NEANF6' in '/home/alex/catkin_ws/src/webrtc_ros/webrtc'
[0:03:56] Cloning into '/home/alex/catkin_ws/src/webrtc_ros/webrtc/src/third_party/_gclient_junit-jar_NEANF6'...
[0:03:57] fatal: remote error: Git repository not found

OSError: [Errno 2] No such file or directory: '/home/alex/catkin_ws/src/webrtc_ros/webrtc/src/third_party/_gclient_junit-jar_NEANF6'
CMake Error at /home/alex/catkin_ws/src/webrtc_ros/webrtc/CMakeLists.txt:20 (message):
  WebRTC source checkout failed
```
Do you have any ideas as to what might be wrong?

wanghongwei123, 2018-01-29:
https://chromium.googlesource.com/external/junit/

freddie-ai, 2018-04-09:
Hi,
Has anyone found a workaround/fix for this?

clement91190, 2018-07-04:
Hi! Also wondering if someone figured out a workaround ? 
Thanks !

chrisui, 2018-10-24:
Also interested.

Cheers!

roehling, 2019-01-16:
The latest version should build on Kinetic

closed 2019-01-16 by roehling