RobotWebTools/webrtc_ros #38: What is the recommended way to build the project?

magnuspyke, 2019-03-29:
What is the recommended way of building this project?

I've cloned the repository into a new catkin workspace.  Running catkin_make fails with the following error:

`Base path: /home/matt/workspace/catkin_ws_webrtc
Source space: /home/matt/workspace/catkin_ws_webrtc/src
Build space: /home/matt/workspace/catkin_ws_webrtc/build
Devel space: /home/matt/workspace/catkin_ws_webrtc/devel
Install space: /home/matt/workspace/catkin_ws_webrtc/install
####
#### Running command: "cmake /home/matt/workspace/catkin_ws_webrtc/src -DCATKIN_DEVEL_PREFIX=/home/matt/workspace/catkin_ws_webrtc/devel -DCMAKE_INSTALL_PREFIX=/home/matt/workspace/catkin_ws_webrtc/install -G Unix Makefiles" in "/home/matt/workspace/catkin_ws_webrtc/build"
####
-- Eclipse version is set to 4.8 (). Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- Using CATKIN_DEVEL_PREFIX: /home/matt/workspace/catkin_ws_webrtc/devel
-- Using CMAKE_PREFIX_PATH: /home/matt/workspace/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/matt/workspace/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.15", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/matt/workspace/catkin_ws_webrtc/build/test_results
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.15") 
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - webrtc (plain cmake)
-- ~~  - webrtc_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/matt/workspace/catkin_ws_webrtc/build/CMakeFiles/CMakeOutput.log".
See also "/home/matt/workspace/catkin_ws_webrtc/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

`
Building with catkin_make_isolated works but pulls in a lot of external dependencies.

vzts, 2019-03-29:
For the external dependencies, it is meant to behave so, because it needs to pull source code from chromium repo, which is a lot(~13GiB).
You may also just leave out the webrtc package with CATKIN_IGNORE and just use the binary from `sudo apt install ros-kinetic-webrtc` if you don't need to configure anything on the webrtc package.

And also, you might want to use [catkin_tools](https://catkin-tools.readthedocs.io/en/latest/history.html#parallel-isolated-catkin-workflow-and-catkin-build) instead of catkin. It's simpler and faster.

magnuspyke, 2019-03-29:
Thanks @vzts for you quick response and suggestions!

closed 2019-03-29 by magnuspyke