RobotWebTools/webrtc_ros #39: Has anyone been able to get this to work over the internet?

pbafff, 2019-06-11:
The streamer only seems to work if both peers are on the same machine or the same local network. The only way I've gotten it to work over the internet is by connecting my desktop to my cloud vm with openvpn. I thought the problem was that stun servers aren't being used so I added stun:stun.l.google.com:19302 to the peer connection config in both webrtc_client.cpp and webrtc_ros.js but no luck.  chrome://webrtc-internals/ shows that iconnectionstatechange is failed but I don't know enough about webrtc to interpret that.

roehling, 2019-06-12:
The problem lies in the NAT traversal, which is almost always required for machines connecting over the Internet. There is a long-standing wishlist bug (#24) for that, but I haven't looked into it yet.

As always, if someone wants to work on it and submit a PR, I'd be more than happy to add such a feature.

pbafff, 2019-06-12:
Do you have any guesses as to what exactly is wrong? I'm going to try to solve this and submit a PR if i get it.

pbafff, 2019-06-13:
Ok so it looks like the streamer only generates host ice candidates for some reason which would explain why internet connections are impossible.