RobotWebTools/webrtc_ros #40: Issues in Firefox

mjsobrep, 2019-10-08:
Just running the demo site, I am able to get everything working in Chrome, but not in Firefox. in Firefox, I get:

>>The character encoding of the HTML document was not declared. The document will render with garbled text in some browser configurations if the document contains characters from outside the US-ASCII range. The character encoding of the page must be declared in the document or in the transfer protocol. viewer

>>Firefox can’t establish a connection to the server at ws://localhost:8080/webrtc. webrtc_ros.js:27:25

>>WebRTC signaling error webrtc_ros.js:36:14

>>WebRTC signaling close webrtc_ros.js:39:14

And nothing shows up in the viewers.

In the terminal, running the node, I get:
```
[ INFO] [1570562248.634517075]: 127.0.0.1:47588: GET /webrtc
[ INFO] [1570562248.634659017]: Handling new WebRTC websocket
[ INFO] [1570562248.636821773]: Creating WebrtcClient
[ WARN] [1570562258.645276080]: Connection broken
[ INFO] [1570562258.645368074]: Destroying Webrtc Client
```
Any thoughts on what could be up? Thanks

roehling, 2019-10-10:
I was able to get Firefox running again, but it took some considerable work: the JavaScript code needed to be updated and Firefox sends at least one invalid ICE candidate during the initial setup phase.

I will probably commit a fix in the next one or two weeks.

mjsobrep, 2019-10-10:
cool, thanks

roehling, 2019-10-14:
I pushed two commits to the develop branch. The changes only affect the `webrtc_ros` package, so you can still use the pre-packaged `webrtc` package (and avoid building gigabytes of source code).

Let me know if that resolves your problem.

mjsobrep, 2019-10-31:
Sorry for being so slow to test this. I'm still having similar problems. In chrome everything still works, but not in Firefox. As you can see in the image below, I am running the new, modified code. But in Firefox connection issues are still coming up. 

[image]

vzts, 2019-11-01:
@mjsobrep Probably because of this:
https://webrtc.org/web-apis/chrome/unified-plan/
Current SDP implementation I believe is based on Plan B format.

roehling, 2019-11-06:
Weirdly enough, I can make it work in Firefox alright. I have added every single remaining patch that I was working on, even though I don't believe they are related. The only thing left that might actually have an impact is [this PR](https://github.com/GT-RAIL/async_web_server_cpp/pull/14) for async_web_server_cpp.

vzts, 2019-11-06:
Oh I missed the console log. Guess the error has to do with websocket code as @roehling said. But in the latest Firefox though, I believe the SDP format issue is still there.

roehling, 2019-11-08:
Another thing that I might have communicated poorly: You do not need to rebuild the `webrtc` package which merely compiles the (external) WebRTC codebase, but you *do* need to rebuild the `webrtc_ros` package that implements the server node. It needs to ignore a malformed SDP line that Firefox insists on sending.

mjsobrep, 2019-11-09:
Ahh, I had not realized that. I pulled down the dev branch of this repo into my `catkin_ws/src`, added a `CATKIN_IGNORE` file to the webrtc package. I also pulled the `apache2_websocket_proxy_workaround` branch of the `async_web_server_cpp` into my `catkin_ws/src`. I re-ran `catkin_make`. Setup a demo video stream to play, ran the `webrtc_ros_server_node` and I have the same result, although with a lot more info getting printed to the terminal now. Is there anything wrong in my approach? It would be useful if anyone else is following this thread to see if this all works for them, maybe my systems are broken in some weird way...

roehling, 2019-11-10:
Your approach is fine. I pretty much did the same things, albeit using catkin_tools to build intead. Regarding your system, I'm using Ubuntu 18.04 with ROS Melodic, and I haven't tested the package with the previous LTS yet. Maybe that's the issue?

mjsobrep, 2019-11-10:
That is possible. I am on kinetic in ubuntu 16

mjsobrep, 2019-12-28:
FInally upgraded to ubuntu 18 with melodic. I'm seeing the same error using the same approach as above. Not the end of the world, chrome works, so I will stick with that for now.

roehling, 2019-12-29:
I'm sorry to hear that. Do you think you could create e.g. a docker image that reproduces the error?

mjsobrep, 2020-04-25:
I'm not sure when this stopped being a problem for me. But it is currently working in firefox. So I am closing this.

closed 2020-04-25 by mjsobrep

roehling, 2020-04-27:
Great news! I'm sorry I didn't have much time to investigate, but I could never reproduce that problem anyway. Hopefully, it stays solved for good.

MoffKalast, 2020-05-22:
Can confirm that this is still a problem with the currently released apt-gettable version for kinetic. I'm seeing the exact same signaling error.

I'll try compiling the current master and see if there are any improvements.

MoffKalast, 2020-05-23:
I've now tested the current master which didn't work. I went back to the original "firefox fix" commit at 309bf57 which wouldn't really compile with the current webrtc release, so I attemped a slightly later 4d7dbeb, which also didn't work in Firefox.

I sincerely doubt this was ever fixed. It is quite puzzling how there could be a signaling error due to websockets not connecting - websockets which obviously work in Firefox without any problems in hundreds of thousands of applications.

rohbotics, 2020-05-24:
Found the culprit here, the WebSocket upgrade request is being responded to with `HTTP/1.0`, but Firefox requires `HTTP/1.1`. This patch fixes it for me: https://github.com/rohbotics/async_web_server_cpp/commit/c994c1211c6f873197caf7991204a201b2bb0503

Reference (under "The WebSocket handshake"): https://developer.mozilla.org/en-US/docs/Web/API/WebSockets_API/Writing_WebSocket_servers