RobotWebTools/webrtc_ros #51: fails to build on ubuntu 20.04

flynneva, 2020-08-30:
after cloning and running `catkin_make_isolated` I am getting this error.

is ubuntu 20.04 noetic not supported?

```
-- Configuring WebRTC build system
ERROR at //webrtc/BUILD.gn:224:12: Unsupported value in libs.
  libs = [ "Foundation.framework" ]
           ^---------------------
Use frameworks to list framework dependencies.
See //BUILD.gn:14:5: which caused the file to be included.
    "//webrtc",
    ^---------
CMake Error at CMakeLists.txt:61 (message):
  cannot prepare WebRTC build system

-- Configuring incomplete, errors occurred!
```

flynneva, 2020-08-31:
looks like the webrtc source is cloning a pretty old commit. going to see if updating that commit hash (or maybe removing it?) fixes my issue

Bitfroest, 2020-09-08:
Hello @flynneva 

did you get it to work?
Any news on a Ubuntu 20.04 focal and ROS noetic port?

roehling, 2020-09-08:
The WebRTC source code is a behemoth, because you need to replicate almost the full stack of the Chromium build system, and the API is only sort of stable, so it is not trivial to switch to a different commit. I plan on upgrading the build to a newer version of WebRTC and then make it work on Ubuntu 20.04 as well, but I'm a bit starved of development time right now.

flynneva, 2020-09-12:
hi @Bitfroest, unfortunately no i was unsuccessful. @roehling  is right....the webrtc code is HUGE and a bit above what I'd be able to debug/help out with.

so far ive tried to update the webrtc hash in the `webrtc_ros/webrtc/build/get_webrtc_source` file to [71d7c8e3cdd61b69d09fc392b93cc8c461168f0d](https://webrtc.googlesource.com/src.git/+/71d7c8e3cdd61b69d09fc392b93cc8c461168f0d) and rerunning the `catkin_make_isolated` command with a clean workspace (removing the old webrtc source).

after doing that, i still am seeing a weird error that im not really sure even where to begin debugging:

```
-- Configuring WebRTC build system
ERROR at //build/timestamp.gni:31:19: Script returned non-zero exit code.
build_timestamp = exec_script(compute_build_timestamp,
                  ^----------
Current dir: /home/flynneva/code/ros/noetic/build_isolated/webrtc/devel/ninja/
Command: python /home/flynneva/code/ros/noetic/src/webrtc_ros/webrtc/build/webrtc/src/build/compute_build_timestamp.py default
Returned 1.
stderr:

Traceback (most recent call last):
  File "/home/flynneva/code/ros/noetic/src/webrtc_ros/webrtc/build/webrtc/src/build/compute_build_timestamp.py", line 126, in <module>
    sys.exit(main())
  File "/home/flynneva/code/ros/noetic/src/webrtc_ros/webrtc/build/webrtc/src/build/compute_build_timestamp.py", line 112, in main
    last_commit_timestamp = int(open(lastchange_file).read())
IOError: [Errno 2] No such file or directory: '/home/flynneva/code/ros/noetic/src/webrtc_ros/webrtc/build/webrtc/src/build/util/LASTCHANGE.committime'

See //base/BUILD.gn:34:1: whence it was imported.
import("//build/timestamp.gni")
^-----------------------------
See //third_party/opus/BUILD.gn:623:7: which caused the file to be included.
      "//base",
      ^-------
CMake Error at CMakeLists.txt:61 (message):
  cannot prepare WebRTC build system

-- Configuring incomplete, errors occurred!
```

GMahmoud, 2020-10-09:
Hello guys :) 

Lately I got the same issue when I tried to update the webrtc commit. 
I checked this missing file `'${PATH_TO_WEBRTC_ROS}/webrtc/build/webrtc/src/build/util/LASTCHANGE.committime'`
and I found this python script `lastchange.py` in the same path `'${PATH_TO_WEBRTC_ROS}/webrtc/build/webrtc/src/build/util'`
When I run it as follow
```  
./lastchange.py LASTCHANGE
```
the missing file is generated :grinning:  
Then I managed to build the webrtc static library by deleting the following flags `linux_use_bundled_binutils=false use_gconf=false` from `prepare_webrtc_build`

I can not give you the insurance that the generated library will work for `webrtc_ros` because it's using a very old commit of webrtc and maybe there some breaking changes. 

Hope that helps 
Cheers

kubja, 2020-11-21:
@GMahmoud I integrated your suggestions to compile under Noetic in pull request #53 . Also, as @roehling suggested I got to upgrade to M86 in that pull request.

GMahmoud, 2020-11-22:
> @GMahmoud I integrated your suggestions to compile under Noetic in pull request #53 . Also, as @roehling suggested I got to upgrade to M86 in that pull request.

Thanks I will take a look at your PR :) soon  and give you my opinion

flynneva, 2020-11-23:
closing due to the great work from @kubja!

closed 2020-11-23 by flynneva

chenYongZu, 2021-01-29:
i meet the problem like you when i build old version of webrtc by the new gn tool, i download old version gn source code and old version ninja source code compilting by myself, then the error disappear.

drummerchin, 2021-08-07:
before version m86, use `frameworks` instead of `libs` in .gn file can solved this problem.
you can find a commit **c7f0dff191 Convert GN libs lists to frameworks** between m85 and m86.

```
8fe932a5a3 (branch-heads/4195) Migrate call/ to webrtc::Mutex.
6287280d64 Migrate audio/ to use webrtc::Mutex
afd1dcbde0 [Adaptation] Move TriggerAdaptationDueToFrameDroppedDueToSize
c3efe1abfa Inclusive language in //rtc_base.
c7f0dff191 Convert GN libs lists to frameworks
d7d2f27b1b [Adaptation] Adaptation holds restrictions
9b79ad33af Roll chromium_revision 33c9ff5246..a9cc4f4732 (785166:785286)
595652ddbd Roll chromium_revision a245a04732..33c9ff5246 (785063:785166)
62032d4592 (branch-heads/4194, branch-heads/4193, branch-heads/4192) Updates rtp_sender_unitests to use separate thread for pacer calls.
2a8932020e Rename sanitizers suppression files.
54d7e1f6f3 Roll chromium_revision a6259aec3c..a245a04732 (784160:785063)
41525d0cc0 red: implement RED with distance 2
```

jacky2050, 2021-12-15:
frameworks = [ "Foundation.framework" ]