RobotWebTools/webrtc_ros #56: catkin_make_isolated error

mpena2099, 2021-01-13:
Hi!

catkin_make_isolated not working on my Ubunu 20.04, ROS noetic.

Steps:

1. `git clone https://github.com/GT-RAIL/async_web_server_cpp.git`

2. 
```
sudo apt-get install libjsoncpp-dev
sudo ln -s /usr/include/jsoncpp/json/ /usr/include/json
```

3. `git clone https://github.com/RobotWebTools/webrtc_ros`

Now,  catkin_make_isolated:

```
(...)
[ 53%] Built target webrtc_ros_server_noncpp11
[ 57%] Linking CXX shared library /home/mauricio/catkin_ws/devel_isolated/webrtc_ros/lib/libwebrtc_ros_server.so
/usr/bin/ld: cannot find -ljsoncpp_object
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/webrtc_ros_server.dir/build.make:591: /home/mauricio/catkin_ws/devel_isolated/webrtc_ros/lib/libwebrtc_ros_server.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:1315: CMakeFiles/webrtc_ros_server.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
<== Failed to process package 'webrtc_ros': 
  Command '['/home/mauricio/catkin_ws/devel_isolated/webrtc/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /home/mauricio/catkin_ws/build_isolated/webrtc_ros && /home/mauricio/catkin_ws/devel_isolated/webrtc/env.sh make -j4 -l4

Command failed, exiting.
```

Any ideia? 

Thanks!
Mauricio

mpena2099, 2021-01-14:
I tried to use the Docker and it seems to be OK.

closed 2021-01-14 by mpena2099

JohnTGZ, 2022-02-28:
As a Ubuntu 20.04 user with ROS Noetic, I have encountered the issue of **/usr/bin/ld: cannot find -ljsoncpp_object** too. One thing that can be done for a successful build is to change the following in `webrtc_ros/CMakeLists.txt`:
```
target_link_libraries(${PROJECT_NAME}_server
  ...
  jsoncpp_object
  ...
)
```
to:
```
target_link_libraries(${PROJECT_NAME}_server
  ...
  jsoncpp_lib
  ...
)
```

This ensures that the correct object is linked to.