RobotWebTools/webrtc_ros #57: Failed to initialize the ADM

mpena2099, 2021-01-22:
Hi!

$ docker run --network=host -t -i webrtc_ros /bin/bash
$ roslaunch webrtc_ros webrtc_ros.launch

And then, on browser, when I click on the image topic, that is what I get on ROS:

```
(...)
auto-starting new master
process[master]: started with pid [448]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to abf3a702-5c5e-11eb-a5ae-5cc9d377293b
process[rosout-1]: started with pid [458]
started core service [/rosout]
process[ice_server_provider-2]: started with pid [465]
process[webrtc_server-3]: started with pid [466]

//
// Fatal error in: ../../../../src/webrtc/build/webrtc/src/media/engine/adm_helpers.cc, line 39
// last system error: 88
// Check failed: 0 == adm->Init() (0 vs. -1)
// Failed to initialize the ADM.[webrtc_server-3] process has died [pid 466, exit code -6, cmd /home/webrtc_ws/devel_isolated/webrtc_ros/lib/webrtc_ros/webrtc_ros_server_node __name:=webrtc_server __log:=/root/.ros/log/abf3a702-5c5e-11eb-a5ae-5cc9d377293b/webrtc_server-3.log].
log file: /root/.ros/log/abf3a702-5c5e-11eb-a5ae-5cc9d377293b/webrtc_server-3*.log
```

Any ideia?

Thanks!
Mauricio

mpena2099, 2021-01-24:
I still think this is a bug but to be able to solve it:
https://www.codeleading.com/article/43242589556/

> The reason is that ubuntu has no sound by default. The way to turn on the root sound:
>         vi /root/.profile add
> 
> pulseaudio --start --log-target=syslog

So, in the docker container:
```
$ pulseaudio --start --log-target=syslog
$ roslaunch webrtc_ros webrtc_ros.launch
```

It still didn't work, but the error has been eliminated.

mpena2099, 2021-01-24:
OK, it is working but not in Firefox, as #40.

roehling, 2021-01-24:
The relevant HTTP/1.1 fix is in the revived async_web_server_cpp package [here](https://github.com/fkie/async_web_server_cpp). The package is released and waiting for sync with the main repository. In the meantime, you can compile it yourself, or take it from [ros-testing](http://wiki.ros.org/TestingRepository).

mpena2099, 2021-01-28:
> The relevant HTTP/1.1 fix is in the revived async_web_server_cpp package [here](https://github.com/fkie/async_web_server_cpp). The package is released and waiting for sync with the main repository. [...]

The "ros-testing" didn't work but compile the package did! :smiley:

yy-tayu, 2022-08-26:
Hi ~
(core_audio_utility_win.cc:672): IsSupported
(core_audio_utility_win.cc:208): C:\WINDOWS\system32\audioses.dll
(core_audio_base_win.cc:168): webrtc::webrtc_win::CoreAudioBase::CoreAudioBase[Output]
(core_audio_base_win.cc:169): Automatic restart: 1
(core_audio_base_win.cc:170): Windows version: 5
(core_audio_utility_win.cc:716): CreateDeviceEnumerator
(core_audio_base_win.cc:188): webrtc::webrtc_win::CoreAudioBase::CoreAudioBase:Registered endpoint notification callback.
(core_audio_output_win.cc:33): webrtc::webrtc_win::CoreAudioOutput::CoreAudioOutput
(core_audio_base_win.cc:168): webrtc::webrtc_win::CoreAudioBase::CoreAudioBase[Input]
(core_audio_base_win.cc:169): Automatic restart: 1
(core_audio_base_win.cc:170): Windows version: 5
(core_audio_utility_win.cc:716): CreateDeviceEnumerator
(core_audio_base_win.cc:188): webrtc::webrtc_win::CoreAudioBase::CoreAudioBase:Registered endpoint notification callback.
(core_audio_input_win.cc:36): webrtc::webrtc_win::CoreAudioInput::CoreAudioInput
(audio_device_module_win.cc:516): webrtc::webrtc_win::CreateWindowsCoreAudioAudioDeviceModuleFromInputAndOutput
(audio_device_module_win.cc:98): webrtc::webrtc_win::`anonymous-namespace'::WindowsAudioDeviceModule::WindowsAudioDeviceModule
(audio_processing_impl.cc:274): Injected APM submodules:
Echo control factory: 0
Echo detector: 0
Capture analyzer: 0
Capture post processor: 0
Render pre processor: 0
(webrtc_voice_engine.cc:263): WebRtcVoiceEngine::WebRtcVoiceEngine
(webrtc_video_engine.cc:588): WebRtcVideoEngine::WebRtcVideoEngine()
(webrtc_voice_engine.cc:285): WebRtcVoiceEngine::Init
(audio_device_module_win.cc:128): webrtc::webrtc_win::`anonymous-namespace'::WindowsAudioDeviceModule::Init

I seem to have a similar problem on Windows.

Any ideia?

Thanks!
Yu

yy-tayu, 2022-08-29:
it is working, I modified the file webrtc\rtc_base\synchronization\sequence_checker.h
`
#define RTC_DCHECK_RUN_ON(x)
#if 0
  webrtc::webrtc_seq_check_impl::SequenceCheckerScope seq_check_scope(x); \
  RTC_DCHECK((x)->IsCurrent()) << webrtc::ExpectationToString(x)          
#endif
#endif  // RTC_BASE_SYNCHRONIZATION_SEQUENCE_CHECKER_H_
`