RobotWebTools/webrtc_ros #65: Build Error

denesh-globotix, 2022-03-31:
**Description**
CMake Error at /home/globotix/catkin_flexa/src/flexa-robot/flexa_static/webrtc_ros/webrtc/CMakeLists.txt:51 (message):
  cannot fetch GN build tool

- Library Version:
- ROS Version: ROS Noetic
- Platform / OS: Ubuntu 20.04

**Steps To Reproduce**

catkin build webrtc_ros

**Expected Behavior**
successful build

**Actual Behavior**
build error due to the GN build tool

wesley-nov, 2022-08-09:
I am also having a build issue using the same command.  It appears that build.ninja file cannot be found, this was after I removed the FATAL_ERROR from the message in the CMakeLists.txt file that refers to the output of the bash script "prepare_webrtc_build".  Perhaps your issue is related to this as well?

varunkotian, 2024-04-08:
I too get the same issue. I think it is because of the updates in the webRTC.

jp-turra, 2024-06-18:
It seams your `python` executable was not found. I had the same issue and i solve it by installing `sudo apt-get install python-is-python3`. 

OBS: If you are familiar with symbolic links you can create one instead of using apt.