RobotWebTools/webrtc_ros #69: Is There 22.04 Humble Version

esamet, 2023-10-24:
I try build 22.04 Humble machine and build stuck on "proccessing" state cant finish.
[image]

ichdich, 2023-11-15:
It took almost 1 hour or so in my case.

esamet, 2023-11-15:
> It took almost 1 hour or so in my case.

OMG thanks, I forgived after 15 minutes, I will try again.

esamet, 2024-03-12:
> Is it working for you? Did you use the develop-ros2 branch?

I try but cannot use in that time

varunkotian, 2024-04-08:
I am also using ROS2 Humble and Ubuntu 22.04.
Did you successfully build?
Which branch did you use? develop or develop-ros2 or master?

esamet, 2024-04-12:
Sorry I really forgot, but as I remember didnt work

tall1, 2024-04-14:
> I am also using ROS2 Humble and Ubuntu 22.04. Did you successfully build? Which branch did you use? develop or develop-ros2 or master?

Try this, it worked for me: [webrtc ros2 humble](https://github.com/fictionlab/web_video_server-ros2?tab=readme-ov-file)

ajay1606, 2025-03-10:
@tall1  Any documentation available for testing [webrtc ros2 humble](https://github.com/fictionlab/web_video_server-ros2?tab=readme-ov-file) .

Like which node to start. 

Appreciate any help.