RobotWebTools/webrtc_ros #70: Compilation Error Ubuntu 20.04 ROS 2: ISO C++ forbids declaration of ‘RTC_DISALLOW_COPY_AND_ASSIGN’ with no type

JLBicho, 2023-12-13:
**Description**
Hello,

I'm trying to compile this package with ROS 2 and I get the error:
```error: ISO C++ forbids declaration of `RTC_DISALLOW_COPY_AND_ASSIGN` with no type [-fpermissive]```

Is there any additional parameter needed in the compilation steps?
Am I missing libraries, dependencies or something like this?

Thank you!

- Library Version: develop-ros2
- ROS Version: ROS 2 Foxy
- Platform / OS: Ubuntu 20.04

**Steps To Reproduce**

1. Clone the repo
2. Compile with colcon build
3. Wait around 20 mins
4. Error regarding RTC_DISALLOW_COPY_AND_ASSIGN

**Expected Behavior**
Compilation succeeds.

**Actual Behavior**
Error during compilation.

```
[Processing: webrtc]omplete] [webrtc:build 61% - 19min 29.3s]
[Processing: webrtc]                                         
[Processing: webrtc]                                         
[Processing: webrtc]                                          
--- stderr: webrtc                                            
Cloning into 'depot_tools'...
WARNING: Your metrics.cfg file was invalid or nonexistent. A new one will be created.
Updating depot_tools...
Updating depot_tools...
Updating depot_tools...
/home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc
CMake Warning:
  Manually-specified variables were not used by the project:

    AMENT_CMAKE_SYMLINK_INSTALL

---
Finished <<< webrtc [21min 20s]
Starting >>> webrtc_ros
--- stderr: webrtc_ros                                    
In file included from /home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc_ros/include/webrtc_ros/webrtc_client.h:7,
                 from /home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc_ros/include/webrtc_ros/webrtc_ros_server.h:6,
                 from /home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc_ros/src/webrtc_ros_server_node.cpp:2:
/home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc_ros/include/webrtc_ros/ros_video_renderer.h:19:48: error: ISO C++ forbids declaration of ‘RTC_DISALLOW_COPY_AND_ASSIGN’ with no type [-fpermissive]
   19 |   RTC_DISALLOW_COPY_AND_ASSIGN(RosVideoRenderer);
      |                                                ^
make[2]: *** [CMakeFiles/webrtc_ros_server_node.dir/build.make:63: CMakeFiles/webrtc_ros_server_node.dir/src/webrtc_ros_server_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/webrtc_ros_server_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< webrtc_ros [10.2s, exited with code 2]

Summary: 2 packages finished [21min 30s]
  1 package failed: webrtc_ros
  2 packages had stderr output: webrtc webrtc_ros
```

JLBicho, 2023-12-13:
I've managed to work around this error by compiling with an additional flag:

``colcon build --cmake-args ' -DCMAKE_CXX_FLAGS=-fpermissive'```

However, now I get a new error:
```
--- stderr: webrtc                                          
/home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc
---
Finished <<< webrtc [7min 31s]
Starting >>> webrtc_ros
--- stderr: webrtc_ros                                    
In file included from /home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc_ros/include/webrtc_ros/webrtc_client.h:7,
                 from /home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc_ros/include/webrtc_ros/webrtc_ros_server.h:6,
                 from /home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc_ros/src/webrtc_ros_server_node.cpp:2:
/home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc_ros/include/webrtc_ros/ros_video_renderer.h:19:48: warning: ISO C++ forbids declaration of ‘RTC_DISALLOW_COPY_AND_ASSIGN’ with no type [-fpermissive]
   19 |   RTC_DISALLOW_COPY_AND_ASSIGN(RosVideoRenderer);
      |                                                ^
/home/jose/ws/webrtc_ros/src/webrtc_ros/webrtc_ros/src/webrtc_client.cpp:8:10: fatal error: webrtc/rtc_base/bind.h: No such file or directory
    8 | #include <webrtc/rtc_base/bind.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/webrtc_ros_server_node.dir/build.make:167: CMakeFiles/webrtc_ros_server_node.dir/src/webrtc_client.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/webrtc_ros_server_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< webrtc_ros [10.0s, exited with code 2]
```

jp-turra, 2024-06-19:
I'm facing the same issue. Looking at installation scripts ( `src/deps/webrtc_ros/webrtc/build/get_webrtc_source` ) i found out the version with hash "93081d594f7efff72958a79251f53731b99e902b" does not have these files. I'm trying to find the closest hash with these files now.

jp-turra, 2024-06-19:
I changed to the same version used in "develop" branch (f183d1d9966b312006e395dc4c270639b35d26de) and rebuild all the project from the ground. It worked for me.