RobotWebTools/webrtc_ros #71: webrtc_ros failing to compile on Ubuntu 22.04, ROS 2 Humble

GTeref, 2024-04-12:
**Description**
Hello,
I am trying to compile this package with ROS2 and I get the following error:

```
CMake Error at CMakeLists.txt:41 (message):
  cannot fetch GN build tool
```

- Library Version: develop-ros2
- ROS Version: ROS 2 Humble
- Platform / OS: Ubuntu 22.04

**Steps To Reproduce**
1. Clone the repo
2. Run ```colcon build```
3. Wait 0.8 seconds
4. CMake Error appears regarding "cannot fetch GN build tool"

**Expected Behavior**
Builds successfully

**Actual Behavior**
CMake error

```
Starting >>> webrtc  
Starting >>> webrtc_ros_msgs
--- stderr: webrtc                                                                    
build/get_gn: line 17: cd: /home/gteref/webrtc_ros/webrtc/build/webrtc/src: No such file or directory
CMake Error at CMakeLists.txt:41 (message):
  cannot fetch GN build tool

---
Failed   <<< webrtc [1.76s, exited with code 1]
Aborted  <<< webrtc_ros_msgs [4.28s]                           

Summary: 0 packages finished [4.59s]
  1 package failed: webrtc
  1 package aborted: webrtc_ros_msgs
  1 package had stderr output: webrtc
  1 package not processed
```

**Note**
I have tried updating the commit hashes to the latest commits in the `build/webrtc` folder, in `get_gn` and `get_webrtc_ros`, but that didn't work. From past issues/pull requests, it seems that updating to a newer version of WebRTC is not as trivial as this.

IliasElabbassi, 2024-05-03:
Hello, I'm facing the same issue to compile on Ubuntu 22.04 and ROS 2 Humble ! Were you able to solve this ?

IliasElabbassi, 2024-05-06:
> ```
> CMake Error at CMakeLists.txt:41 (message):
>   cannot fetch GN build tool
> ```

Try adding execute privilege to the scripts within `~/webrtc_ws/src/webrtc/build`

```
chmod +x ~/webrtc_ws/src/webrtc/build/get_*
chmod +x ~/webrtc_ws/src/webrtc/build/prepare_webrtc_build
```

This might solve this problem, but it will not solve all the errors that might occur when building the package.

EricVoll, 2024-07-12:
I made some progress on this:

in `webrtc_ros/webrtc/build/get_gn` you have to remove `python` in L3 and replace `python` with `python3` on L29. 
Additionally I downloaded `gn` and `ninja` manually and added it to the path. Then deleted the webrtc and depot_tools folders and now it has been building for about 5 minutes...

ajay1606, 2025-02-20:
@EricVoll would you please help me with correct branch to build in Ubuntu 22.04, humble.

adlimited, 2025-04-02:
> **Description** Hello, I am trying to compile this package with ROS2 and I get the following error:
>
> ```
> CMake Error at CMakeLists.txt:41 (message):
>   cannot fetch GN build tool
> ```
>
> * Library [...]

Did you manage to solve it?